Copilot-Language/copilot #589: `copilot-theorem`: Produce a counterexample when the What4 backend falsifies a property

RyanGlScott, 2025-01-29:
**Description**

Currently, the [`Copilot.Theorem.What4.prove`](https://hackage.haskell.org/package/copilot-theorem-4.2/docs/Copilot-Theorem-What4.html#v:prove) function returns a list of results, where each result contains a `SatResult` that describes whether a property is `Valid`, `Invalid`, or `Unknown`. The `Invalid` result has the limitation that it does not give any information about a specific counterexample that could drive Copilot into falsifying the property, however. This makes it challenging to interpret what the results of `prove` mean.

It would be helpful if `Copilot.Theorem.What4` could offer an API to prove or disprove a property such that disproven properties come with a concrete counterexample. This counterexample information could then 

**Type**

* Feature: Add counterexample capabilities to the What4 backend in `copilot-theorem`.

**Additional context**

None.

**Requester**

* Ryan Scott (Galois).

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

Introduce a new function to `Copilot.Theorem.What4` that mirrors the type signature of `prove`, except that it returns a variant of `SatResult` where the `Invalid` equivalent encodes counterexample information. `copilot-theorem` users can then interpret the results of the counterexample in Copilot specifications.  

**Desired result**

Introduce a new function to `Copilot.Theorem.What4` that mirrors the type signature of `prove`, except that it returns a variant of `SatResult` where the `Invalid` equivalent encodes counterexample information. `copilot-theorem` users can then interpret the results of the counterexample in Copilot specifications.  

**Proposed solution**

Introduce a new `prove' :: Solver -> Spec -> IO [(Name, SatResult' CounterExample)]` function (names subject to change during review), where `SatResult'` is defined to be:

```hs
data SatResult' = Valid' | Invalid' CounterExample | Unknown'
```

And `CounterExample` records enough information about a concrete counterexample such that a Copilot user could display it.

**Further notes**

None.

ivanperez-keera, 2025-02-13:
Technical Lead: Issue scheduled for fixing in Copilot 4.3.

Fix assigned to: @RyanGlScott .

RyanGlScott, 2025-02-24:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2025-02-28:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://github.com/Copilot-Language/copilot/runs/37925988037
  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile runs an example included in the PR that demonstrates the new feature by printing the results of proving some combination of valid and invalid properties, for which counterexamples are produced when applicable. I have checked by visual inspection that the results are produced, and that the message is well formatted and informative enough:
    ```Dockerfile
    FROM ubuntu:focal
    
    ENV DEBIAN_FRONTEND=noninteractive
    RUN apt-get update
    
    RUN apt-get install --yes \
          libz-dev \
          git \
          curl \
          gcc \
          g++ \
          make \
          libgmp3-dev  \
          pkg-config \
          z3
    
    RUN mkdir -p $HOME/.ghcup/bin
    RUN curl https://downloads.haskell.org/~ghcup/0.1.19.2/x86_64-linux-ghcup-0.1.19.2 -o $HOME/.ghcup/bin/ghcup
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    
    SHELL ["/bin/bash", "-c"]
    
    RUN ghcup install ghc 9.10
    RUN ghcup install cabal 3.2
    RUN ghcup set ghc 9.10
    RUN cabal update
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
      && cabal v1-sandbox init \
      && cabal v1-install alex happy --constraint='happy <= 2' \
      && cabal v1-install $NAME/copilot**/ \
      && cabal v1-exec -- runhaskell $NAME/copilot/examples/what4/ArithmeticCounterExamples.hs \
      && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/GaloisInc/copilot-1" -e "NAME=copilot-1" -e "COMMIT=9861071718850fb8ef85e533a6c0d6d083f69094" -it copilot-verify-589
    ```
- [X] Implementation is documented. Details:
  All new top-level definitions include haddock documentations, as well as the internals of the code and the examples.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  A new example is introduced to demonstrate the feature.
- [X] Required version bumps are evaluated. Details:
  Bump required; the new feature changes the types of some value constructors in `copilot-theorem`, thus affecting the public API.

ivanperez-keera, 2025-02-28:
Change Manager: Implementation ready to be merged.

commit f39b9daf (2025-02-28): Merge branch 'develop-copilot-theorem-counterexamples-take-three'. Close #589.

**Description**

Currently, the Copilot.Theorem.What4.prove function returns a list of results,
where each result contains a `SatResult` that describes whether a property is
`Valid`, `Invalid`, or `Unknown`. The `Invalid` result has the limitation that
it does not give any information about a specific counterexample that could
drive Copilot into falsifying the property, however. This makes it challenging
to interpret what the results of `prove` mean.

It would be helpful if `Copilot.Theorem.What4` could offer an AP [... cut]
PR #595 (2025-02-28): `copilot-theorem`: Add function to produce counterexamples for invalid properties. Refs #589.
Currently, the `Copilot.Theorem.What4.prove` function returns a list of results, where each result contains a `SatResult` that describes whether a property is `Valid`, `Invalid`, or `Unknown`. The `Invalid` result has the limitation that it does not give any information about a specific counterexample that could drive Copilot into falsifying the property, however. This makes it challenging to interpret what the results of prove mean.

This introduces a new `proveWithCounterExample` function to `Copilo [... cut]
closed 2025-02-28 by ivanperez-keera