Copilot-Language/copilot #590: `copilot`: Update README to reflect compatibility with GHC 9.10

ivanperez-keera, 2025-01-30:
**Description**

Copilot is currently able to work with GHC 9.10, but the README still indicates that it is compatible with versions up to 9.8. That needs to be updated.

**Type**

- Maintenace: Update README to reflect compatibility with new version of the compiler.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

The README indicates that Copilot is compatible with versions of GHC up to 9.10 (as of this writing).

**Desired result**

The README indicates that Copilot is compatible with versions of GHC up to 9.10 (as of this writing).

**Proposed solution**

Update mentions of GHC versions in the README to 9.10 instead of 9.8.

**Further notes**

None.

ivanperez-keera, 2025-01-31:
Technical Lead: Issue scheduled for fixing in Copilot 4.3.

Fix assigned to: @estherdconrad.

estherdconrad, 2025-02-03:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2025-02-03:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/274059521
  - [X] The solution proposed produces the expected result. Details:
    I have checked that the README now indicates the 9.10 version of GHC. This has been carried by visual inspection: the change is to the documentation, so there's no easy, automated way to check that this is correct.
- [X] Implementation is documented. Details:
   No updates needed; change is to the documentation.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
   No updates needed; change is to the documentation.
- [X] Author is internal or has provided signed CLA.
- [X] Required version bumps are evaluated. Details:
   Bump not required; change is to the documentation.

ivanperez-keera, 2025-02-03:
Change Manager: Implementation ready to be merged.

commit e234f1b0 (2025-02-03): Merge branch 'develop-readme-ghc9.10'. Close #590.

**Description**

Copilot is currently able to work with GHC 9.10, but the README still indicates
that it is compatible with versions up to 9.8. That needs to be updated.

**Type**

- Maintenace: Update README to reflect compatibility with new version of the
  compiler.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

The README indicates that Copilot is compatible with versions of GHC up to 9.10
(as of this writing).

**Solution implemented**

 [... cut]
closed 2025-02-03 by ivanperez-keera