Copilot-Language/copilot #612: Fix build status icon in README

ivanperez-keera, 2025-03-29:
**Description**

The banner for the build status in the README doesn't show up right. Github has cached the result but it's not providing the resulting image, so we see "Build Status" instead.

**Type**

- Bug: Image in README renders incorrectly.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

The only way to reproduce this bug is by viewing the README on github.

**Expected result**

The image for the build status banner renders correctly when the README is read on Github.

**Desired result**

The image for the build status banner renders correctly when the README is read on Github.

**Proposed solution**

Replace banner URL with https://api.travis-ci.com/Copilot-Language/copilot.svg?branch=master.

**Further notes**

None.

ivanperez-keera, 2025-04-12:
Technical Lead: Issue scheduled for fixing in Copilot 4.4.

Fix assigned to: @kaBeech .

kaBeech, 2025-04-29:
Implementor: Solution implemented, review requested.

commit 0d99a80b (2025-05-01): Merge branch 'fix-build-status-icon'. Close #612.
closed 2025-05-01 by ivanperez-keera