Copilot-Language/copilot #622: `copilot-verifier`: Include in mainline

ivanperez-keera, 2025-05-16:
**Description**

The `copilot-verifier` was developed outside of this repo, but there's interest in having all Copilot projects and libraries merged into a main repo for ease of management.

**Type**

- Management: files need to be moved between repos.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

The last state of the `copilot-verifier` is included in this repository, disregarding history, and adjusting the changelog to make the old
issues refer to `copilot-verifier#<ISSUE_NUM>` so that they are not interpreted as related to the main Copilot repo.

**Desired result**

The last state of the `copilot-verifier` is included in this repository, disregarding history, and adjusting the changelog to make the old
issues refer to `copilot-verifier#<ISSUE_NUM>` so that they are not interpreted as related to the main Copilot repo.

**Proposed solution**

Copy last state of `copilot-verifier` as subdirectory of main repo.

Adjust Changelog in `copilot-verifier` to make older issues point to correct repo.

**Further notes**

None.

RyanGlScott, 2025-05-16:
I briefly looked into this, and I have some questions about how to handle certain aspects of the `copilot-verifier` repo:

* Should I list myself as the maintainer in `copilot-verifier.cabal`, or should I list you (as is done for other `copilot` subpackages)?
* Do we still want to include the [`Dockerfile`](https://github.com/Copilot-Language/copilot-verifier/blob/cea9f8e924fe8f50cdbffd15ea7e00d7f73f1d8a/Dockerfile) from the `copilot-verifier` repo?
* There are a variety of examples under the [`copilot-verifier/examples`](https://github.com/Copilot-Language/copilot-verifier/tree/cea9f8e924fe8f50cdbffd15ea7e00d7f73f1d8a/copilot-verifier/examples/Copilot/Verifier) subdirectory. Should I try to move these examples to `copilot/examples` (where the bulk of the Copilot examples are currently stored), or should I keep this examples within the `copilot-verifier` subdirectory?

ivanperez-keera, 2025-05-17:
Technical Lead: Issue scheduled for fixing in Copilot 4.5.

Fix assigned to: @RyanGlScott .

ivanperez-keera, 2025-05-17:
@RyanGlScott Just copy it as is, and we can make the changes later (separately).

Include the example and the `Dockerfile` where they are.

RyanGlScott, 2025-06-20:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2025-06-21:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/275542308
  - [X] The solution proposed produces the expected result. Details:
    The following dockerfile checks out the latest version of `copilot-verifier`, v4.4, and compares it against the version being merged, ignoring differences in the CHANGELOG (which is expected to be different), checking that there are no other differences, after which it prints the message "Success": 
    ```Dockerfile
    FROM ubuntu:noble
    
    ENV DEBIAN_FRONTEND=noninteractive
    RUN apt-get update
    
    RUN apt-get install --yes git
    RUN git config --global advice.detachedHead false
    
    CMD git clone $REPO $NAME \
        && cd $NAME \
        && git checkout $COMMIT \
        && cd .. \
        && git clone -b v4.4 https://github.com/Copilot-Language/copilot-verifier copilot-verifier-original \
        && diff --exclude 'CHANGELOG' -r $NAME/copilot-verifier copilot-verifier-original/copilot-verifier \
        && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e REPO=https://github.com/GaloisInc/copilot-1 -e NAME=copilot -e COMMIT=9e0dcac323f523277b1bdcb74869d823b6ad5324 copilot-verify-622
    ```
- [X] Implementation is documented. Details:
   CHANGELOG updated to make old issues point to original repo. No other changes needed.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
   No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed; change includes existing package without affecting API.

ivanperez-keera, 2025-06-21:
Change Manager: Implementation ready to be merged.

PR #628 (2025-06-21): `copilot-verifier`: Include in mainline `copilot` repo. Refs #622.
The `copilot-verifier` package was originally developed in an external repo (https://github.com/Copilot-Language/copilot-verifier), but there is interest in having all Copilot projects and libraries merged into the main https://github.com/Copilot-Language/copilot repo for ease of management.

This includes the last state of the old `copilot-verifier` repo into the main Copilot repo as a new subdirectory, disregarding the commit history. In a separate commit, this also adjusts the `copilot-verifier` changelog to make the old iss [... cut]