Copilot-Language/copilot #657: `copilot`: Add FPGA example

ivanperez-keera, 2025-08-21:
**Description**

We need an example that demonstrates how to work with FPGAs using Copilot. This will help users start with `copilot-bluespec`, and also help validate our work.

**Type**

- Feature: New example.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

**Expected result**

The `copilot` package includes an example that shows how to use `copilot-bluespec`.

**Desired result**

The `copilot` package includes an example that shows how to use `copilot-bluespec`.

**Proposed solution**

Create README under `copilot/examples/fpga`, in a folder for the specific example.

Add any necessary auxiliary files.

**Further notes**

None.

ivanperez-keera, 2025-08-21:
Technical Lead: Confirmed that the issue should be addressed.

sukhmankkahlon, 2025-08-21:
I can work on this.

ivanperez-keera, 2025-09-06:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/276236949
  - [X] The solution proposed produces the expected result. Details:
    The PR implements a manual, so it's been evaluated by visual inspection. The manual is well structure, the steps to install the tools and implement the example are clear, and the solution looks complete and takes the user all the way to having an example running on an FPGA.
- [X] Implementation is documented. Details:
  The change includes a manual, which is well documented.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  The change is to the examples.
- [X] Author is internal or has provided signed CLA.
  The author accomplished this work while being a NASA intern.
- [X] Required version bumps are evaluated. Details:
  Bump not required; the change is to the examples.

ivanperez-keera, 2025-09-06:
Change Manager: Implementation ready to be merged.

commit f4f8bf4d (2025-09-06): Merge branch 'develop-examples-fpga-helloworld'. Close #657.

**Description**

We need an example that demonstrates how to work with FPGAs using
Copilot. This will help users start with `copilot-bluespec`, and also
help validate our work.

**Type**

- Feature: New example.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

**Expected result**

The `copilot` package includes an example that shows how to use
`copilot-bluespec`.

**Solution implemented**

Create README under `copilot/examples/fpga`, in a folder for the
specific example.

Add any nec [... cut]
closed 2025-09-06 by ivanperez-keera